yizhou-wang/RODNet #2: ConfMap generation details

jb892, 2021-01-13:
Hi yizhou,

I'm reading through your code. But I'm confused about the details of confMap generation process. At this line of code, can you explain how sigma is calculated? What's the meaning of `confmap_length[class_name]` and `range_grid[rng_idx]`?
https://github.com/yizhou-wang/RODNet/blob/969cad6f08b8957b26bc16f86ac4e835d1294050/rodnet/core/confidence_map/generate.py#L36

Also, why you time 2 to `(rng_idx - i)` at the following line? 
https://github.com/yizhou-wang/RODNet/blob/0505a4d8f416f34b01384b130ecc5f1122e5d2f5/rodnet/core/confidence_map/generate.py#L44

Thanks!

yizhou-wang, 2021-01-18:
The pattern of the reflection in RF images is usually an ellipse shape. Times 2 here is empirically defined to better "mimic" the radar signal reflection.

jb892, 2021-01-18:
OK, cool! Thanks!

closed 2021-01-18 by jb892

zhengzangw, 2021-02-01:
I am still confused about how variance is computed. In the paper, it says "the variance is related to the object class and scale information", while in the code, `sigma` is calculated by

https://github.com/yizhou-wang/RODNet/blob/969cad6f08b8957b26bc16f86ac4e835d1294050/rodnet/core/confidence_map/generate.py#L36

As I understand, `sigma` is the class-specific variance, and `confmap_sigma` is a hyper-parameter for variance per class. My question is:

- Is there a physical meaning or intuition behind `np.arctan(confmap_length[class_name] / (2 * range_grid[rng_idx]))`
- Is `confmap_sigmas`, `confmap_sigmas_interval` and `confmap_length` all hyper-parameters choosed via experiments?

Thanks in advance.