yizhou-wang/RODNet #6: cann't create the test/.pkl file

sunpengfei1122, 2021-01-29:
Hi, for my understanding, the "prepare.py" will generate training data(image and radar) and testing data(only radar). But the code seems to try to generate both even when I deal with the testing data. then it will raise the error(no image folder). correct me if I am wrong. Thanks.

yizhou-wang, 2021-01-31:
Yes, the prepare_data.py will generate pkl files including data paths and ConfMaps. For the testing data without images, you can just remove the image paths from the saved dictionary.

sunpengliang, 2021-01-31:
> Yes, the prepare_data.py will generate pkl files including data paths and ConfMaps. For the testing data without images, you can just remove the image paths from the saved dictionary.

so should I modify the prepare_data.py file when generating pkl files of testing data?

closed 2021-02-01 by yizhou-wang

chencaichang, 2021-02-04:
Hi, have you solved this problem? How did you deal with the  prepare_data.py to get the test/.pkl  file? Thank you very much for your reply.

sunpengliang, 2021-02-04:
> Hi, have you solved this problem? How did you deal with the prepare_data.py to get the test/.pkl file? Thank you very much for your reply.
I think you need to modify the scirpt,  in the challenge, RGB images and annotations for the testing set are not provided, just comment them out to make it works.

chencaichang, 2021-02-04:
> Hi, have you solved this problem? How did you deal with the prepare_data.py to get the test/.pkl file? Thank you very much for your reply.
> I think you need to modify the scirpt,  in the challenge, [...]
Thank you very much for your reply. But I‘m not sure how to modify it? Can you specify which file to modify?

chencaichang, 2021-02-05:
> Yes, the prepare_data.py will generate pkl files including data paths and ConfMaps. For the testing data without images, you can just remove the image paths from the saved dictionary.

Hi, I tried to process the test data and removed the two functions :load_anno_txt()、generate_confmaps(), and the test./pkl obtained was only about 1.6M. Please correct me if I am doing something wrong. Thank you very much.

yizhou-wang, 2021-02-05:
@chencaichang That's should be correct without the confmaps included.

Strongly20, 2021-10-12:
> > 是的，prepare_data.py 将生成 pkl 文件，包括数据路径和 ConfMap。对于没有图像的测试数据，您可以从保存的字典中删除图像路径。
> 
> 你好，我尝试处理测试数据，去掉了load_anno_txt()、generate_confmaps()两个函数，得到的test./pkl只有1.6M左右。如果我做错了什么，请纠正我。非常感谢。

您好，请问您可以把您修改过的prepare_data.py给我发一份吗？我的邮箱是1142919491@qq.com 。非常感谢！

d28081439, 2021-11-16:
> > > 是的，prepare_data.py 将生成 pkl 文件，包括数据路径和 ConfMap。对于没有图像的测试数据，您可以从保存的字典中删除图像路径。
> > 
> > 
> > 你好，我尝试处理测试数据，去掉了load_anno_txt()、generate_confmaps()两个函数，得到的test./pkl只有1.6M左右。如果我做错了什么，请纠正我。非常感谢。
> 
> 您好，请问您可以把您修改过的prepare_data.py给我发一份吗？我的邮箱是[1142919491@qq.com](mailto:1142919491@qq.com) 。非常感谢！

您好,請問你有成功獲得使用prepare_data.py獲得test.pkl了嗎,能否寄一份給我？ 我的mail是 d28081439@gmail.com ,非常感謝

yizhou-wang, 2022-01-27:
In case someone still cannot generate `test/*.pkl` successfully, I just updated the script [here](https://github.com/yizhou-wang/RODNet/commit/7c3fd6f996f1f3699e5131b0312a770c8896fcb2) to handle the situation when camera images and annotations are not available for the ROD2021 test set. Hope it helps!