yizhou-wang/RODNet #7: cdc模型，测试验证集的时候，有的类别得分输出概率大于1

yuzehui1996, 2021-01-30:
您好！
感谢您的源码。我在测试cdc模型的时候，有的类别得分输出大于1，导致评测AP和AR的时候会报错：Confidence score of obj1 should not be smaller than obj2. ""obj1['score'] = %s, obj2['score'] = %s"

后来我查看cdc模型和hgwi模型的decode结构时，发现二者有差异：
cdc模型decode最后没有加prelu激活：
x = self.prelu(self.convt1(x))  # (B, 256, W/4, 16, 16) -> (B, 128, W/2, 32, 32)
        x = self.prelu(self.convt2(x))  # (B, 128, W/2, 32, 32) -> (B, 64, W, 64, 64)
        x = self.convt3(x)  # (B, 64, W, 64, 64) -> (B, 3, W, 128, 128)
        return x

而hgwi模型最后有加prelu激活：
x = self.prelu(self.convt1(x + x3))  # (B, 256, W/4, 16, 16) -> (B, 128, W/2, 32, 32)
        x = self.prelu(self.conv1(x))
        x = self.prelu(self.convt2(x + x2))  # (B, 128, W/2, 32, 32) -> (B, 64, W, 64, 64)
        x = self.prelu(self.conv2(x))
        x = self.prelu(self.convt3(x + x1))  # (B, 64, W, 64, 64) -> (B, 3, W, 128, 128)
        x = self.prelu(self.conv3(x))
        return x

但是原论文的模型示意图上面，所有模型decode模块的最后都加了prelu激活，不知这是否是一个code bug？
感谢！

sunpengfei1122, 2021-01-30:
你好，我test阶段都有点问题，没有办法生成test.pkl.你是改了prepare.py的代码吗

yizhou-wang, 2021-01-31:
The output of the confmap is greater than 1 is because you are using MSE loss. You can use BCE loss instead to make sure all the values in the confmap are between 0 and 1.  

https://github.com/yizhou-wang/RODNet/blob/969cad6f08b8957b26bc16f86ac4e835d1294050/tools/train.py#L133

As for the test pkl file generation, please take a look at #6 to see if you met the same issue.

closed 2021-02-01 by yizhou-wang