yizhou-wang/RODNet #8: The location of implementation for CRF algorithm

zhengzangw, 2021-01-31:
In your paper, you mention that CRF Annotations are generated from both RGB images and RAMaps Snippet by CRF algorithm. However, after reading through the code, I couldn't find where the camera image is used. It seems that the confmap is only generated by the radar data.

It would be of great help if you can help point out where the CRF algorithm is implemented. Thanks in advance.

yizhou-wang, 2021-01-31:
Sorry, we are not planning to release the CRF code since it contains a large number of different sections, which are very difficult to implement and integrate. For the ROD2021 Challenge, you can either directly use our provided annotations, or propose a new algorithm for better annotation. However, a new annotation algorithm is not encouraged since our target is to propose more advanced networks for radar-only object detection.

zhengzangw, 2021-02-01:
Thanks for your explanation.

closed 2021-02-01 by zhengzangw