yizhou-wang/RODNet #10: Multi-GPUs train？

yuzehui1996, 2021-02-05:
Hi，
Have you ever tried muti-gpus training? I simply add DataParallel but the AP and AR are lower than the training with single gpu.

Thanks!

yizhou-wang, 2021-02-05:
Sorry, we didn't try multiple GPUs before. But the performance should not downgrade using multiple GPUs. You may need to tune some parameters or monitor the loss during the training.

yuzehui1996, 2021-02-07:
Thanks for your reply!
I have trained the houglass model(HG) with multi-gpus. When I test the checkpoints and set batch-size from 1 to 4, the AP and AR downgrade greatly. Will the accuracy decrease due to the BatchNormalization layer during multi-gpus training?
BTW, I use 4 gpu cards for training.

yizhou-wang, 2021-02-09:
I double-checked the inference code. Seems the batch size is hardcoded as 1. We will fix this bug in the future. But for now, you can just use `batch_size = 1` in the testing phase.