yizhou-wang/RODNet #11: Access to the validation script

ArthurOuaknine, 2021-02-12:
Hi Yizhou.

I would like to train a RODNet from scratch. To do so, it could be great to have validation scores to select the best models and analyse the training process. Referring to your paper, I'm talking about the method you used to compute the AP and AR scores using the OLS thresholds between the selected points from the predicted ConfMaps and the ground truth.
Your training script mentions a validation function that seems to have this purpose here:
https://github.com/yizhou-wang/RODNet/blob/969cad6f08b8957b26bc16f86ac4e835d1294050/tools/train.py#L232

Maybe I have missed something in your code to compute the performances of the model.
If yes,, could you indicate me where I can find this information. If not, could you please integrate this function in the repo?

Thanks for your help, have a great day.
Best.
Arthur

yizhou-wang, 2021-02-12:
Hi @ArthurOuaknine, 

The evaluation code and functions are included in our devkit package: https://github.com/yizhou-wang/cruw-devkit. The detailed evaluation tutorial can be found here: https://github.com/yizhou-wang/cruw-devkit/blob/master/tutorials/cruw_devkit_tutorial_rod2021.ipynb line [17]. Hope it helps!

ArthurOuaknine, 2021-02-13:
Thank you very much for the devkit, I didn't know there was one!
There is everything I need in it. You can close this issue.

closed 2021-02-13 by yizhou-wang