yizhou-wang/RODNet #13: Confidence greater than 1(BUG)

Huang-Chuan, 2021-02-25:
RODNet predicts a class with conf > 1, is it common?

File "/media/chuan/MyPassport/deep_learning/cruw-devkit/cruw/eval/metrics.py", line 21, in get_ols_btw_objects
    "obj1['score'] = %s, obj2['score'] = %s" % (obj1['score'], obj2['score']))
TypeError: Confidence score of obj1 should not be smaller than obj2. obj1['score'] = 1.0, obj2['score'] = 1.055537462234497

yizhou-wang, 2021-02-25:
The output of the confmap is greater than 1 is because you are using MSE loss. You can use BCE loss instead to make sure all the values in the confmap are between 0 and 1.

https://github.com/yizhou-wang/RODNet/blob/969cad6f08b8957b26bc16f86ac4e835d1294050/tools/train.py#L133

[Duplicate of #7 ]

Huang-Chuan, 2021-02-25:
Thank you very much!

closed 2021-02-25 by Huang-Chuan