yizhou-wang/RODNet #14: CRDataset.py ignores trailling data

Huang-Chuan, 2021-02-27:
The CRDataset will not include some data files at the back of a training folder.
Say Folder **2019_04_30_MLMS001** has 900 frames, with a **win_size** of 16 and **stride_size** of 8, the CRDataset only gives a length of 111, and the last data it includes corresponds to frame 895. 
```
if __name__ == "__main__":
    args = parse_args()
    args.config = 'configs/config_demo.py'
    config_dict = load_configs_from_file(args.config)
    dataset = CRUW(data_root="../ROD2021", sensor_config_name='sensor_config_rod2021')
    crdata_train = CRDataset(data_dir='data/', dataset=dataset, config_dict=config_dict, split='test',
                                 noise_channel=args.use_noise_channel)
    print(len(crdata_train))

    dataloader = DataLoader(crdata_train, 1, shuffle=False, num_workers=0, collate_fn=cr_collate)
    for iter, data_dict in enumerate(dataloader):
        #print(iter)
        if iter == len(dataloader) - 1:
            print(len(data_dict['image_paths'][0]))
            print(data_dict['image_paths'][0])`
```

outputs:
```
111
['../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000880.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000881.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000882.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000883.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000884.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000885.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000886.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000887.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000888.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000889.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000890.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000891.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000892.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000893.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000894.jpg', '../ROD2021/sequences/test/2019_04_30_MLMS001/IMAGES_0/0000000895.jpg']
```

yizhou-wang, 2021-02-27:
Yes, you are right. If the length of the last remaining frames is not sufficient to form a whole snippet, we drop it for convenience. It will drop a maximum of one snippet in each data sequence, which is not critical. But you are free to add them back during developing your algorithm.

Huang-Chuan, 2021-02-27:
Thanks. Any suggestions on how to add them back? Got stuck on this before submiting the results.

yizhou-wang, 2021-02-27:
You can try to infer from the last 16 frames and merge the results of the overlapped frames.

Huang-Chuan, 2021-02-28:
Thank you very much.

closed 2021-02-28 by Huang-Chuan