yizhou-wang/RODNet #15: Question with images in test data set

kathy-lee, 2021-03-02:
Hi yizhou,

thanks for your code sharing of RODNet! I have a small question about test data set: I reorganize the data files as you showed the data file structure in CRUW devkit, but I got an error during running 'perpare_data.py', because it cannot find image file in test subdir. But I thought in test stage there will only be radar data and no image file. Could you help me to make this clear? Thanks!

chunguangqu, 2021-03-04:
I encountered the same problem as you, have you found a solution?Thank you

kathy-lee, 2021-03-04:
Hi @chunguangqu ,
after trying I found 'prepare_data.py' can run through by: choose samples from training set and just let them be test samples. The RODNet does only use radar data as network input, the images are only for visualization use. So I think it's not an error, but need to change some snippets for doing test without image data.

yizhou-wang, 2021-03-05:
Duplicate of #6 

To prepare `test.pkl` file, you can just comment the image paths out. 
@kathy-lee Yes, you are right. the RODNet does not require images as input during testing.

closed 2021-03-06 by yizhou-wang