yizhou-wang/RODNet #16: Error on radar_data

songsanling, 2021-03-02:
File "tools/train.py", line 174 in <module>
    for iter, data_dict in enumerate(dataloader):
  File "C:\anaconda3\envs\rodnet\lib\site-packages\torch\utils\data\dataloader.py", line 435, in __next__
    data = self._next_data()
  File "C:\anaconda3\envs\rodnet\lib\site-packages\torch\utils\data\dataloader.py", line 475, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "C:\anaconda3\envs\rodnet\lib\site-packages\torch\utils\data\_utils\fetch.py", line 47, in fetch
    return self.collate_fn(data)
  File "c:\rodnet\rodnet\datasets\collate_functions.py", line 48, in cr_collate
    return {key: cr_collate([d[key] for d in batch]) for key in elem}
  File "c:\rodnet\rodnet\datasets\collate_functions.py", line 48, in <dictcomp>
    return {key: cr_collate([d[key] for d in batch]) for key in elem}
  File "c:\rodnet\rodnet\datasets\collate_functions.py", line 48, in <listcomp>
    return {key: cr_collate([d[key] for d in batch]) for key in elem}
KeyError: 'radar_data'

yizhou-wang, 2021-03-05:
Please give a more detailed explanation, so that I can understand what issue you are facing. Thanks!

closed 2021-03-17 by yizhou-wang

songsanling, 2021-03-29:
Sorry, that is my cuda cudnn issue. Already solved. Thanks.