yizhou-wang/RODNet #19: Questions about the format of the dataset

yaoshanliang, 2021-04-19:
I have downloaded the dataset from your website. The format is shown below.
[image]

In file cruw_devkit_tutorial_rod2021.ipynb, the format is
[image]

Do I need to transfer the downloaded dataset into the format in the ipynb file?
Is the transfer script already provided?

yaoshanliang, 2021-04-27:
@yizhou-wang

yizhou-wang, 2021-05-06:
Hi @yaoshanliang, what you need to do is to put the files according to the file structure shown in the ipynb file. You do not need to use any scripts.

closed 2022-01-27 by yizhou-wang