yizhou-wang/RODNet #30: 关于训练模型的问题

Strongly20, 2021-09-22:
Traceback (most recent call last):
  File "tools/train.py", line 181, in <module>
    data = data_dict['radar_data']
KeyError: 'radar_data'

yizhou-wang, 2022-01-27:
No detailed error information is mentioned for this issue. Close this issue for now. If you have further questions with more detailed error descriptions, I would reopen this issue.

closed 2022-01-27 by yizhou-wang

Strongly20, 2022-01-27:
这个问题已经解决了，谢谢！

YoushaaMurhij, 2022-02-11:
@Strongly20 , Can you please provide the solution?

puyoulingdi, 2022-07-15:
> 这个问题已经解决了，谢谢！

@Strongly20  Hi, could you please assist to tell me how to solve this problem, I met the same problem, thank you!

Strongly20, 2022-11-17:
> > 这个问题已经解决了，谢谢！
> 
> @Strongly20 你好，请问你是怎么解决这个问题的，我遇到了同样的问题，谢谢！

这里我并没有详细记录，我记得当时在debug的过程中发现代码运行到CRDataset.py中有一行路径会导致停止运行，我将该路径重写以后代码就运行成功了。您可以参考以下！