yizhou-wang/RODNet #32: about the data input of the RODNet model

blueskych, 2021-10-11:
Hi,
I want to test the model on a  customer datasets, which is the range,  azimuth and reflection(rcs) value. And i have trained the RODNet model based on the CRUW dataset . So how can I use my own datasets to test the RODNet model ? 
Thanks for you help.

blueskych, 2021-10-12:
> Hi, I want to test the model on a customer datasets, which is the range, azimuth and reflection(rcs) value. And i have trained the RODNet model based on the CRUW dataset . So how can I use my own datasets to test the RODNet model ? Thanks for you help.

Or  the value rcs and azimuth represent the real and imaginary parts ？

yizhou-wang, 2022-01-27:
We haven't tested our model on other kinds of radar hardware. But as long as you have the same data format and annotations as the CRUW dataset, RODNet is supposed to handle it. 

For the data representation, please refer to https://github.com/yizhou-wang/cruw-devkit/issues/10.

closed 2022-01-27 by yizhou-wang