yizhou-wang/RODNet #36: 关于ex_evaluate_rod2021.py

Strongly20, 2022-01-25:
请问代码中submit_dir指的是什么路径？

yizhou-wang, 2022-01-26:
This is the evaluation code for the ROD2021 Challenge. `submit_dir` is your detection results folder. The format is the same as the ROD2021 Challenge submission.

Strongly20, 2022-01-26:
您的意思是按照这种格式新建一个文件夹，并将检测结果的.txt改名后存入吗？我在annotations的同级目录下按上面的方法创建了这样的路径，但是仍然报错FileNotFoundError: [Errno 2] No such file or directory: '/mnt/workspace/RODNet/data_raw/sequences/test/IMAGES_0'

[image]

yizhou-wang, 2022-01-27:
Does this error appear while running `ex_evaluate_rod2021.py`? Which script are you working with?

Strongly20, 2022-01-27:
是的，我之前将结果的.txt改成了每个场景文件夹的名字后放在了整理好格式的数据集目录下。我在运行的是cruw-devkit/scripts/ex_evaluate_rod2021.py然后出现了这条报错。

yizhou-wang, 2022-01-27:
I have pushed a new commit [here](https://github.com/yizhou-wang/cruw-devkit/commit/c4d5906600b33fcd3ea4ebd60591dfb1a6280a0c) to handle this issue. As long as you have the radar chirp data folder available, this script should work now.

Strongly20, 2022-01-28:
非常感谢您的耐心回复！但是我发现修改完之后还是报错FileNotFoundError: [Errno 2] No such file or directory: '/mnt/workspace/RODNet/data_raw/sequences/test/RADAR_RA_H'，我怀疑也许是我的路径设置有问题，图中是我的文件目录设置，evaluate是我新加的一个文件夹，我将测试结果中每个文件夹中的rod_res.txt改成了文件夹的名字后存入evaluate文件夹中，并且将ex_evaluate_rod2021.py中的两个路径设置成submit_dir = '/home/ncut/Desktop/workspace/RODNet/data_raw/evaluate' 和truth_dir = '/home/ncut/Desktop/workspace/RODNet/data_raw/annotations'，请问我的这些操作都是正确的吗？
[image]

yizhou-wang, 2022-01-29:
This error is because the evaluation code is using radar data to determine the frame number of each sequence. You should provide the correct path to the radar data.

closed 2022-02-19 by yizhou-wang

Strongly20, 2022-03-02:
非常感谢您的问题，之前的报错我已经解决了。但是还有一个问题就是，测试时检测结果文件夹存在但是测试集没有真实标注，truth_dir应该选择什么路径呢？