yizhou-wang/RODNet #39: Camera-Radar Fusion code sharing

Davy4fg, 2022-02-23:
Hi Yizhou and RODNet team members,

Thanks for the great work and sharing. 

Is it possible to share the camera-radar fusion part to generate the annotation?

yizhou-wang, 2022-02-23:
Sorry, we are not planning to release CRF code because it requires a lot of different inputs and is difficult to be released well-organized. However, you can find the camera part [here](https://github.com/yizhou-wang/MOLoc3D), and the radar part is just classical radar CFAR detection. You can then implement the fusion part yourself using the formula mentioned in the RODNet paper. 

Besides, we may release the code for another annotation method mentioned in [this paper](https://openaccess.thecvf.com/content/CVPR2021W/WAD/html/Wang_Rethinking_of_Radars_Role_A_Camera-Radar_Dataset_and_Systematic_Annotator_CVPRW_2021_paper.html) in the future. 

Hope it helps!

closed 2022-02-23 by yizhou-wang

Davy4fg, 2022-02-24:
Thanks for the comments and looking forward to the new radar object annotation release.