yizhou-wang/RODNet #40: Definition of radar data

Davy4fg, 2022-02-28:
Hi Yizhou,

Thanks for the precious radar, camera and ground truth data for training.
I didn't find the definition of how the radar data(.npy in RADAR_RA_H) produced. 
Could you please give a brief work flow from the raw adc data to it?
For example:
adc data --> range fft --> doppler fft --> angle fft --> save to .npy file
What are the number of Tx/Rx antenna used, number of chirp, number of range/ doppler bin in raw data?

yizhou-wang, 2022-02-28:
- Here is our radar data processing:
adc data --> range fft --> angle fft --> save to .npy file (mentioned in RODNet paper)

- What are the number of Tx/Rx antenna used, number of chirp, number of range/ doppler bin in raw data?
Most of the statistics are mentioned in the RODNet paper. # of range bins: 128.

- In the ROD2021 dataset, we only released 4 chirps since the full dataset is too large to be used in the challenge. For fair comparisons on the leaderboard, we encourage you to start with 4 chirps. We may consider releasing the 8 chirps later.

yizhou-wang, 2022-03-08:
We are using AWR1843BOOST with 2Tx and 4Rx. The lower-level radar data preprocessing can be found here: https://github.com/Xiangyu-Gao/mmWave-radar-signal-processing-and-microDoppler-classification.

closed 2022-03-15 by yizhou-wang