yizhou-wang/RODNet #42: about images in test datasets

xiaoyangxxx, 2022-03-01:
Hi，Yizhou,

Thank you for providing data to train and test
When I was reading the paper, I found that it tested radar data with some poorly lighted (nighttime) images.  But when I downloaded CRUW, I found that there is no such data.I think it is probably in the test set, however there are no visual images to prove my conjecture. Recently I want to do some visual and radar fusion tasks and need these night images very much. I wonder if you can share this part of the data.

Once again, thank you sincerely!

yizhou-wang, 2022-03-08:
We were not planning to release the RGB image data in the testing set. However, upon your request, I think we can release that images for the Vision-Hard scenario only. Please download them [here](https://drive.google.com/file/d/1952gYiGTMHc24gq8nJEdMjlB9EbJJkym/view?usp=sharing).

xiaoyangxxx, 2022-03-16:
Thanks again for everything!