yizhou-wang/RODNet #43: 关于test.py的疑惑

slj1997, 2022-03-06:
您好，非常感谢您的指导
当我运行test.py的时候，我有一些疑惑。
首先，我已经在prepare_date.py程序运行过程中，成功生成了训练集中RGB图像+射频图像的.pkl文件，并且生成sequences/text/文件中射频图像的.pkl文件。

但是当我运行test.py程序过程中，我的程序停止在170行左右 data = data_dict['radar_data']      image_paths = data_dict['image_paths'][0]     seq_name = data_dict['seq_names'][0]  。data_dict这个字典中并没有所需要的key。

此外，检查前文中加载的测试文件.pkl中，并没有发现关于image_paths的属性。考虑是因为生成测试.pkl文件中，并没有引用RGB文件的路径。请问我的理解是否是正确的？

我所加载config文件为：config_rodnet_cdc_win16.py
sensor_config文件为：sensor_config_rod2021
data_dir文件为：prepare_date.py中生成的射频图像的.pkl。
checkpoint文件为：训练过程中最后一轮生成的.pkl文件

感谢您提供的宝贵意见！！

yizhou-wang, 2022-03-08:
Thanks for your interest! 

Our method does not need RGB images during the inference stage. The `image_paths` in `test.py` is just for visualization purposes. To make it runnable, you can remove the variables related to RGB images in `test.py`.

yizhou-wang, 2022-03-15:
Hi, I have updated `test.py` script to fix this issue.

closed 2022-03-15 by yizhou-wang

slj1997, 2022-03-18:
感谢
1