yizhou-wang/RODNet #45: Compile pip install -e . with error

rong-xue, 2022-03-16:
pip install -e reports an error. According to the online solution is not solved, help!The following screenshots of the error are attached, without the screenshots of Requirement already satisfied and Warning!
[image]

yizhou-wang, 2022-03-16:
Could you please provide the full error message? I cannot know the reason for your error from this screenshot.

rong-xue, 2022-03-16:
我把terminal的内容保存为word文档和mark文档了，您可以看看！还需要什么，我再给你，麻烦您了！
[RODNet.docx](https://github.com/yizhou-wang/RODNet/files/8260266/RODNet.docx)
[截图.md](https://github.com/yizhou-wang/RODNet/files/8260268/default.md)

rong-xue, 2022-03-16:
我将terminal的内容及自己的一些步骤复制到mark文件中了，看起来可能比word会好一些！我自己目前也不知是什么问题，不知道如何解决，因此多麻烦您了！
[rodnet.md](https://github.com/yizhou-wang/RODNet/files/8260553/rodnet.md)
[image]

yizhou-wang, 2022-03-17:
Are you using the TDC module or not? If you are using TDC, please use CUDA=10.1 and pytorch=1.4 as mentioned in readme.

rong-xue, 2022-03-17:
I followed the steps and chose to not use TDC. Enter the following statement.
conda install pytorch torchvision cudatoolkit=10.1 -c pytorch  # if not using TDC.
In this case, I just changed cudatoolkit=10.1 to 10.2.
Do I need to use TDC, rebuild the environment and try again. Previously I chose not to use TDC, but only distinguished it by the above statement, I don't know if there is any other way to distinguish it.
[image]
.

yizhou-wang, 2022-03-17:
From the log you sent me through email, you were compiling TDC. Please make sure you follow the steps. 

BTW, we previously did not set up the environment in Windows, but that should not be a problem.

rong-xue, 2022-03-17:
Okay, thank you! I'll rebuild the environment and try again with the corresponding version.

yizhou-wang, 2022-04-04:
I think the configurations and environment are mentioned in the readme. We only tested on ubuntu 16, 18, and 20.