yizhou-wang/RODNet #53: tools/test.py出现keyError

lightzwc, 2022-04-20:
你好，我运行这个命令的时候，
python tools/test.py --config configs/config_rodnet_cdc_win16.py --data_dir ./data/result/ --checkpoint checkpoints/rodnet-cdc-win16-wobg-20220419-105954/epoch_100_final.pkl --res_dir results/
（data/result文件夹是我生成的train和test的pkl文件）
出现这个问题，
Traceback (most recent call last):
  File "tools/test.py", line 184, in <module>
    data = data_dict['radar_data']
KeyError: 'radar_data'
python tools/test.py --config configs/config_rodnet_cdc_win16.py --data_dir ./data/result/ --checkpoint checkpoints/rodnet-cdc-win16-wobg-20220419-105954/epoch_100_final.pkl --res_dir results/

rong-xue, 2022-04-20:
您好，我看您进行到test.py这一步，可以请问一下python tools/prepare_dataset/prepare_data.py这一命令怎么输入，我尝试了很多次都失败了，目前没有弄懂什么原因！
[image]

lightzwc, 2022-04-20:
> 您好，我看您进行到test.py这一步，可以请问一下python tools/prepare_dataset/prepare_data.py这一命令怎么输入，我尝试了很多次都失败了，目前没有弄懂什么原因！ [image]

你好，请参考
python tools/prepare_dataset/prepare_data.py --config configs/config_rodnet_cdc_win16.py --data_root data_root/ --split train,test --out_data_dir data/result

lightzwc, 2022-04-26:
路径问题，已解决

closed 2022-04-26 by lightzwc

xuezhen5267, 2022-05-04:
能问一下楼上，问题如何解决的？我也有同样的报错，路径哪里有问题？

lightzwc, 2022-05-04:
[image]

Heng-Dickens, 2023-03-13:
请问有修改这个文件的详细信息吗？我也遇到了同样的问题不知道如何解决，我不清楚你指的npy文件是哪些文件？

ZhongKang97, 2023-04-16:
[image]
should not end with "/" ,or it will cause wrong path when loading data from the the pkl file ,like this 
[image]
[image]