yizhou-wang/RODNet #55: if i choose config.py with mnet to test, Value error in chrip_ops.py:c0,c1,c2 =chirp.shape too many values to unpack(expected 3)

Linda0111, 2022-04-29:


closed 2022-10-09 by yizhou-wang

hopenessQAQ, 2024-04-13:
@yizhou-wang    I have a similar problem. Do you know how to solve it

hopenessQAQ, 2024-04-13:
@yizhou-wang  I have a similar problem. Do you know how to solve it ？          Since I want to see the training effect, set the n epoch to 5                                                                                                                                                                                                **********************                                                                                                                                                                                    
 python tools/test.py --config configs/config_rodnet_hg1v2_win16_mnet.py  --data_dir data/datadir --checkpoint checkpoints/rodnet-hg1v2-win16-mnet-20240413-023955/epoch_05_final.pkl   --res_dir results/
Warning: DCN modules are not correctly imported!
Warning: DCN modules are not correctly imported!
Warning: DCN modules are not correctly imported!
Warning: DCN modules are not correctly imported!
Building model ... ({'type': 'HGv2', 'name': 'rodnet-hg1v2-win16-mnet', 'max_dets': 20, 'peak_thres': 0.4, 'ols_thres': 0.3, 'stacked_num': 1, 'mnet_cfg': (4, 32), 'dcn': False})
['2019_04_09_BMS1000', '2019_04_09_BMS1001', '2019_04_09_BMS1002', '2019_04_09_CMS1002', '2019_04_09_PMS1000', '2019_04_09_PMS1001', '2019_04_09_PMS2000', '2019_04_09_PMS3000', '2019_04_30_MLMS000', '2019_04_30_MLMS001', '2019_04_30_MLMS002', '2019_04_30_PBMS002', '2019_04_30_PBMS003', '2019_04_30_PCMS001', '2019_04_30_PM2S003', '2019_04_30_PM2S004', '2019_05_09_BM1S008', '2019_05_09_CM1S004', '2019_05_09_MLMS003', '2019_05_09_PBMS004', '2019_05_09_PCMS002', '2019_05_23_PM1S012', '2019_05_23_PM1S013', '2019_05_23_PM1S014', '2019_05_23_PM1S015', '2019_05_23_PM2S011', '2019_05_29_BCMS000', '2019_05_29_BM1S016', '2019_05_29_BM1S017', '2019_05_29_MLMS006', '2019_05_29_PBMS007', '2019_05_29_PCMS005', '2019_05_29_PM2S015', '2019_05_29_PM3S000', '2019_09_29_ONRD001', '2019_09_29_ONRD002', '2019_09_29_ONRD005', '2019_09_29_ONRD006', '2019_09_29_ONRD011', '2019_09_29_ONRD013']
2019_04_09_BMS1000
100%|█████████████████████████████████████████████| 1/1 [00:00<00:00,  2.42it/s]
Length of testing data: 221
Traceback (most recent call last):
  File "tools/test.py", line 234, in <module>
    radar_input = chirp_amp(data.numpy()[0, :, i, :, :], radar_configs['data_type'])
  File "/home/sd-robot/radar_yhw/RODNet/rodnet/core/radar_processing/chirp_ops.py", line 11, in chirp_amp
    c0, c1, c2 = chirp.shape
ValueError: too many values to unpack (expected 3)
**********************