yizhou-wang/RODNet #58: [ERROR]    pip install -e .

chenhengwei1999, 2022-07-26:
**When I execute pip install -e. to install rodnet, The following error occurred:**

    ERROR: Command errored out with exit status 1:
     command: /home/chw/anaconda3/envs/rodnet/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/chw/projects/mmradar_signal_process/RODNet/setup.py'"'"'; __file__='"'"'/home/chw/projects/mmradar_signal_process/RODNet/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /home/chw/projects/mmradar_signal_process/RODNet/
    Complete output (58 lines):
    running develop
    running egg_info
    writing rodnet.egg-info/PKG-INFO
    writing dependency_links to rodnet.egg-info/dependency_links.txt
    writing requirements to rodnet.egg-info/requires.txt
    writing top-level names to rodnet.egg-info/top_level.txt
    reading manifest file 'rodnet.egg-info/SOURCES.txt'
    adding license file 'LICENSE'
    writing manifest file 'rodnet.egg-info/SOURCES.txt'
    running build_ext
    building 'rodnet.ops.dcn.deform_conv_2d_cuda' extension
    gcc -pthread -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Wno-deprecated-declarations -fPIC -DWITH_CUDA -I/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/include -I/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/include/TH -I/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/include/THC -I:/usr/local/cuda-11.4/include -I/home/chw/anaconda3/envs/rodnet/include/python3.6m -c rodnet/ops/dcn/src/deform_conv_2d_cuda.cpp -o build/temp.linux-x86_64-3.6/rodnet/ops/dcn/src/deform_conv_2d_cuda.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=deform_conv_2d_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++11
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    /home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/command/easy_install.py:159: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
      EasyInstallDeprecationWarning,
    /home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      setuptools.SetuptoolsDeprecationWarning,
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/chw/projects/mmradar_signal_process/RODNet/setup.py", line 115, in <module>
        zip_safe=False
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
        return distutils.core.setup(**attrs)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/command/develop.py", line 34, in run
        self.install_for_development()
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/command/develop.py", line 114, in install_for_development
        self.run_command('build_ext')
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 79, in run
        _build_ext.run(self)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/command/build_ext.py", line 339, in run
        self.build_extensions()
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 372, in build_extensions
        build_ext.build_extensions(self)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
        self._build_extensions_serial()
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
        self.build_extension(ext)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/setuptools/command/build_ext.py", line 202, in build_extension
        _build_ext.build_extension(self, ext)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
        depends=ext.depends)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/distutils/ccompiler.py", line 574, in compile
        self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 288, in unix_wrap_compile
        "'-fPIC'"] + cflags + _get_cuda_arch_flags(cflags)
      File "/home/chw/anaconda3/envs/rodnet/lib/python3.6/site-packages/torch/utils/cpp_extension.py", line 1027, in _get_cuda_arch_flags
        raise ValueError("Unknown CUDA arch ({}) or GPU not supported".format(arch))
    ValueError: Unknown CUDA arch (8.6) or GPU not supported
> ERROR: Command errored out with exit status 1: /home/chw/anaconda3/envs/rodnet/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/chw/projects/mmradar_signal_process/RODNet/setup.py'"'"'; __file__='"'"'/home/chw/projects/mmradar_signal_process/RODNet/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.

**The final root cause is: <ValueError: Unknown CUDA arch (8.6) or GPU not supported>.** 
**But I checked my environment, cuda version and pytorch version is consistent with those in your tutorial.**

[image]

[image]

**My gpu information is as below:**

[image]

**May I ask where the problem may be? Thank you in advance for your answer.**

chenhengwei1999, 2022-07-26:
By the way, my NVIDIA GPU info and ubuntu info is as below:

[image]

[image]

MaximilianArnold, 2022-08-11:
Hi,

if you are interested. I put a Docker version on it. But you need CUDA 10.X for it (same for the original repo)
https://github.com/MaximilianArnold/RODNet-Docker

yizhou-wang, 2022-10-09:
@chenhengwei1999 Can you double-check if your cuda is properly installed or version is correct?

Thanks @MaximilianArnold ! I will test it later. @chenhengwei1999 , you can also try @MaximilianArnold 's docker image for software setup.

RouChan, 2022-11-03:
Hey, I encountered the same problem under Python 3.7 with Pytorch1.4. Have you solved it?

Judy-996, 2024-07-19:
I'm encountering the same issue as you mentioned, so have you resolved it now?

WHATABLOCK, 2025-03-31:
> I'm encountering the same issue as you mentioned, so have you resolved it now?

delete setup.py and rename setup_wo_tdc.py as setup.py