yizhou-wang/RODNet #63: 关于DeformConvPack3D找不到的问题

Strongly20, 2022-09-07:
您好！非常感谢您的代码分享！
我在尝试使用TDC的过程中运行train.py时出现了如下报错：
Traceback (most recent call last):
  File "/home/ncut/Desktop/workspace/RODNet-master/tools/train.py", line 169, in <module>
    dcn=config_dict['model_cfg']['dcn']).cuda()
  File "/home/ncut/Desktop/workspace/RODNet-master/rodnet/models/rodnet_hgwi_v2.py", line 22, in __init__
    self.conv_op = DeformConvPack3D
NameError: name 'DeformConvPack3D' is not defined
在调试的过程中确实路径可以找到'DeformConvPack3D'这个类，但是我发现在deform_conv_3d.py中 from . import deform_conv_3d_cuda这一行显示有问题，请问是否需要编译之类的操作呢？
期待您的回复！再次感谢！

Strongly20, 2022-09-13:
我在将部分局部路径修改为绝对路径后问题得到了解决，谢谢！

closed 2022-09-13 by Strongly20

wwang624, 2022-11-17:
您好，我也遇到了同样问题，请问具体是改的哪些路径呢？

Strongly20, 2022-11-17:
我记得我做过的操作有：在rodnet/models/rodnet_hgwi_v2.py下加入了一行sys.path.append("# deform_conv_3d.py的路径")；将from .... import DeformConvPack3D那行from后面的部分路径写全；重新安装了rodnet（pip install -e .)。大概是这些操作，然后就运行成功了，您可以试一下！