yizhou-wang/RODNet #68: test.py : FileNotFoundError

rong-xue, 2022-10-18:
I get the following error when I try to run tools/test.py:
`warning: loading calibration data failed
Building model ... ({'type': 'CDC', 'name': 'rodnet-cdc-win16', 'max_dets': 20, 'peak_thres': 0.3, 'ols_thres': 0.3})
Traceback (most recent call last):
  File "tools/test.py", line 145, in <module>
    seq_names = sorted(os.listdir(os.path.join(data_root, dataset_configs['test']['subdir'])))
FileNotFoundError: [WinError 3] 系统找不到指定的路径。: '/mnt/disk1/CRUW/ROD2021/sequences\\test'`
This is the command I am running:
`python tools/test.py --config configs/config_rodnet_cdc_win16.py --data_dir data/result --checkpoint checkpoints/checkpoint/rodnet-cdc-win16-20221016-200310/epoch_10_final.pkl --res_dir resu
lts/`

rong-xue, 2022-10-18:
I modified the config file to have the paths on my computer, and I am not getting the following output:
`Building model ... ({'type': 'CDC', 'name': 'rodnet-cdc-win16', 'max_dets': 20, 'peak_thres': 0.3, 'ols_thres': 0.3})
['2019_05_28_CM1S013', '2019_05_28_MLMS005', '2019_05_28_PBMS006', '2019_05_28_PCMS004', '2019_05_28_PM2S012', '2019_05_28_PM2S014', '2019_09_18_ONRD004', '2019_09_18_ONRD009', '2019_09_29_ONRD012', '2019_10_13_ONRD048']
2019_05_28_CM1S013
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████
██████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 998.64it/s]
Length of testing data: 111
Traceback (most recent call last):
  File "tools/test.py", line 185, in <module>
    data = data_dict['radar_data']
KeyError: 'radar_data'`
This is the command I am operating about configs/config_rodnet_cdc_win16.py:
`   # dataset_name='ROD2021',
    # base_root="/mnt/disk1/CRUW/ROD2021",
    # data_root="/mnt/disk1/CRUW/ROD2021/sequences",
    # anno_root="/mnt/disk1/CRUW/ROD2021/annotations",

    dataset_name='ROD2021',
    base_root="data_root",                 # data_root is my dataset
    data_root="data_root/sequences",
    anno_root="data_root/annotations",`
I print data_dict and the output is as follows:
`print(data_dict)      # {'status': False, 'seq_names': ['2019_05_28_CM1S013'], 'image_paths': [[]]}`
[image]

mikepark95, 2022-10-24:
I am also running into same error. Any updates?

KeepSake0, 2022-12-27:
> I modified the config file to have the paths on my computer, and I am not getting the following output:
> 
> ```
> Building model ... ({'type': 'CDC', 'name': 'rodnet-cdc-win16', 'max_dets': 20, 'peak_thres': 0.3, 'ols_thres': 0.3})
> ['2019_05_28_CM1S013', '2019_05_28_MLMS005', '2019_05_28_PBMS006', '2019_05_28_PCMS004', '2019_05_28_PM2S012', '2019_05_28_PM2S014', '2019_09_18_ONRD004', '2019_09_18_ONRD009', '2019_09_29_ONRD012', '2019_10_13_ONRD048']
> 2019_05_28_CM1S013
> 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████
> ██████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 998.64it/s]
> Length of testing data: 111
> Traceback (most recent call last):
>   File "tools/test.py", line 185, in <module>
>     data = data_dict['radar_data']
> KeyError: 'radar_data'
> ```
> 
> This is the command I am operating about configs/config_rodnet_cdc_win16.py:
> 
> ```
>     # dataset_name='ROD2021',
>     # base_root="/mnt/disk1/CRUW/ROD2021",
>     # data_root="/mnt/disk1/CRUW/ROD2021/sequences",
>     # anno_root="/mnt/disk1/CRUW/ROD2021/annotations",
> 
>     dataset_name='ROD2021',
>     base_root="data_root",                 # data_root is my dataset folder name
>     data_root="data_root/sequences",
>     anno_root="data_root/annotations",
> ```
> 
> I print data_dict and the output is as follows: `print(data_dict) # {'status': False, 'seq_names': ['2019_05_28_CM1S013'], 'image_paths': [[]]}` [image]

Hi，did you solve this problem?

braundeniz, 2023-06-06:
Hey all!
 After 3 days of searching for errors i found out, that the problem are the missing corresponding testing images to the radardata.

If you look up the structure, the images are missing, which is not a huge problem by itself. More the missing annotations for the testing dataset. Without these, you cant evaluate your trainings.

9527huang, 2023-06-11:
> 

你好，我遇到过你的问题，现在全部解决了。根据你的提示内容中“/mnt/disk1/CRUW/ROD2021/sequences\\test'”，很明显你在Windows系统下跑的代码，因为使用os.path.join的时候会出现“\\”，所以最好是在linux系统下跑代码。
Hi, I had your problem and now it's all solved. According to the content of your prompt "/mnt/disk1/CRUW/ROD2021/sequences\\\test'", it's obvious that you are running the code under Windows, because using os.path.join comes up "\\", so it is better to run the code under linux system.

因为我参考的是深蓝课程，当你切换到linux系统后，你需要到config文件夹里面修改配置文件的路径，改成你的自己的情况，可以参考“KeepSake0”的回答
Because I refer to the deep blue course, when you switch to linux system, you need to go to the config folder to change the path of the configuration file to your own case, you can refer to "KeepSake0" answer

现在回复一下KeepSake0
Now reply to KeepSake0

9527huang, 2023-06-11:
> Hi，did you solve this problem?

你好，我参考了这位朋友的回答：并且安装他的图片中的内容进行了修改
https://github.com/yizhou-wang/RODNet/issues/53
Hello, I refer to this friend's answer: and install his picture in the content of the modified
https://github.com/yizhou-wang/RODNet/issues/53

You, I refer to this friend's answer: and install his picture in the content of the modified
https://github.com/yizhou-wang/RODNet/issues/53
我找了下对应py文件修改位置是：/media/huang/新加卷1/BAIDUYUNDOWNLOAD/radar_homework/RODNet/rodnet/datasets/CRDataset.py
的第143行：
This is my own modification.
elif radar_configs['data_type'] == 'ROD2021':
                if isinstance(chirp_id, int):
                    radar_npy_win = np.zeros((self.win_size, ramap_rsize, ramap_asize, 2), dtype=np.float32)
                    for idx, frameid in enumerate(
                            range(data_id, data_id + self.win_size * self.step, self.step)):
                        radar_path=os.path.join('/media/huang/new-added-volume1/BAIDUYUNDOWNLOAD/radar_homework/RODNet',radar_paths[frameid][chirp_id])
                        radar_npy_win[idx, :, :, :] = np.load(radar_path)
                        If image_paths is not None:
                            data_dict['image_paths'].append(image_paths[frameid])
                        If image_paths is None:
                            data_dict['image_paths']=None

hopenessQAQ, 2024-04-12:
> > > I modified the config file to have the paths on my computer, and I am not getting the following output:
> > > ```
> > > Building model ... ({'type': 'CDC', 'name': 'rodnet-cdc-win16', 'max_dets': 20, 'peak_thres': 0.3, 'ols_thres': 0.3})
> > > ['2019_05_28_CM1S013', '2019_05_28_MLMS005', '2019_05_28_PBMS006', '2019_05_28_PCMS004', '2019_05_28_PM2S012', '2019_05_28_PM2S014', '2019_09_18_ONRD004', '2019_09_18_ONRD009', '2019_09_29_ONRD012', '2019_10_13_ONRD048']
> > > 2019_05_28_CM1S013
> > > 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████
> > > ██████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 998.64it/s]
> > > Length of testing data: 111
> > > Traceback (most recent call last):
> > >   File "tools/test.py", line 185, in <module>
> > >     data = data_dict['radar_data']
> > > KeyError: 'radar_data'
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > This is the command I am operating about configs/config_rodnet_cdc_win16.py:
> > > ```
> > >     # dataset_name='ROD2021',
> > >     # base_root="/mnt/disk1/CRUW/ROD2021",
> > >     # data_root="/mnt/disk1/CRUW/ROD2021/sequences",
> > >     # anno_root="/mnt/disk1/CRUW/ROD2021/annotations",
> > > 
> > >     dataset_name='ROD2021',
> > >     base_root="data_root",                 # data_root is my dataset folder name
> > >     data_root="data_root/sequences",
> > >     anno_root="data_root/annotations",
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > I print data_dict and the output is as follows: `print(data_dict) # {'status': False, 'seq_names': ['2019_05_28_CM1S013'], 'image_paths': [[]]}` [image]
> > 
> > 
> > Hi，did you solve this problem?
> 
> 你好，我参考了这位朋友的回答：并且安装他的图片中的内容进行了修改 #53 Hello, I refer to this friend's answer: and install his picture in the content of the modified #53
> 
> You, I refer to this friend's answer: and install his picture in the content of the modified #53 我找了下对应py文件修改位置是：/media/huang/新加卷1/BAIDUYUNDOWNLOAD/radar_homework/RODNet/rodnet/datasets/CRDataset.py 的第143行： This is my own modification. elif radar_configs['data_type'] == 'ROD2021': if isinstance(chirp_id, int): radar_npy_win = np.zeros((self.win_size, ramap_rsize, ramap_asize, 2), dtype=np.float32) for idx, frameid in enumerate( range(data_id, data_id + self.win_size * self.step, self.step)): radar_path=os.path.join('/media/huang/new-added-volume1/BAIDUYUNDOWNLOAD/radar_homework/RODNet',radar_paths[frameid][chirp_id]) radar_npy_win[idx, :, :, :] = np.load(radar_path) If image_paths is not None: data_dict['image_paths'].append(image_paths[frameid]) If image_paths is None: data_dict['image_paths']=None
> 
> 然后在tool/test.py中修改了，我自己都改成了train，因为这样就可以看到图片了，改成test是没有图的 ![image](https://private-user-images.githubusercontent.com/24982296/244910637-085e47b1-c999-4645-96ca-204c0d4bb366.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5NDM3MDEsIm5iZiI6MTcxMjk0MzQwMSwicGF0aCI6Ii8yNDk4MjI5Ni8yNDQ5MTA2MzctMDg1ZTQ3YjEtYzk5OS00NjQ1LTk2Y2EtMjA0YzBkNGJiMzY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEyVDE3MzY0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3ZmE3NDY4ODZkZWYwNjdmMzkzOGYxMmIyNzhhYzFkZTI5YzhhNGQwNjFkMzQwZjUxNjI4NjhlYzE1MzYwMjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KeIR97UpqOpkj7RDb18AdYDN--0DbaYSRi8tOZ0YXbM)
> 
> 我还在这里进行了修改：改成了store_false ![image](https://private-user-images.githubusercontent.com/24982296/244910681-4b4177ac-f14d-4996-8f54-2023fd334c83.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5NDM3MDEsIm5iZiI6MTcxMjk0MzQwMSwicGF0aCI6Ii8yNDk4MjI5Ni8yNDQ5MTA2ODEtNGI0MTc3YWMtZjE0ZC00OTk2LThmNTQtMjAyM2ZkMzM0YzgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEyVDE3MzY0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk2MjQwYTA4YTAyNjE5YWZlMGIxZGE4YjE5ZjVjZWM3ODdmZWM4MzBhNGQxMGViZmQwNTM3ZjYzZjYzMmNkZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qApm-6Wlq68uwch0diXujX27pONTuzxeZBDgGGiq4IE)
> 
> 我的运行指令是： (datadir)是我自己运行preparedata.py后的文件目录，里面是pkl文件 ![image](https://private-user-images.githubusercontent.com/24982296/244910797-4d757b59-9b92-420c-b471-828b5c14ca20.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI5NDM3MDEsIm5iZiI6MTcxMjk0MzQwMSwicGF0aCI6Ii8yNDk4MjI5Ni8yNDQ5MTA3OTctNGQ3NTdiNTktOWI5Mi00MjBjLWI0NzEtODI4YjVjMTRjYTIwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEyVDE3MzY0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4NzRlNGE0NGJiZjBmMGNiNDY5YjE0OGI1ZWRmNzI5MThkNTdkZGFmYWRlNTUwNzJiMzE0MTM2MzRmOGI3MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bHPO5aW5VodAssfFFjsOOuvTD7h_cJYvTxnxllkJo80)
> 
[... cut ...] N0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEyVDE3MzY0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjMzc5ZmRiNTc5MWE4YjY1YmNhODM5MDA3YjMyNmMwZDk4ZGUwN2JhMTAxYTY1NjEyZjUyMjRhZGZiYzgwNjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7mZ1V6gNmK5brIdYJH0JNCwwuub_zbfu0XVYerJX6cc) 希望有帮助

“这里的train和test差不多，我没放train了”          为什么我按照作者提供的脚本在annotations文件中没有test文件呢?而且这个test中的数据在[ROD2021 dataset](https://www.cruwdataset.org/download#h.mxc4upuvacso).  也没有提到  ，