yizhou-wang/RODNet #73: Questions about the test set and test.py

Heng-Dickens, 2023-03-16:
Hi, I recently read your paper  RODNet, and found it very inspiring. But something went wrong during the test.
[image]
When the program runs to line 192, TypeError:'NoneType' object is not subscribable appears. Then I check the CRDataset.py and found that when the annotation information does not exist, data_dict['anno'] is none. I think this is the cause of the error above. While in the CRUW dataset, there is no annotation file provided. So I wonder how to run the test.py. Or is there something else wrong with me that prevents the program from running?
looking forward to an answer.

promaxj, 2023-03-16:
Maybe need to confirm the placement of the dataset.

ref: https://github.com/yizhou-wang/RODNet#prepare-data-for-rodnet

zhchen17, 2023-08-03:
In test.py, image_path is used several times, in fact this doesn't exist, and from the composition of the dataset, there is only radar data in the test set, no image data, so it doesn't work.